hydronics2/2019-easy-bee-counter #1: Suggestion...

owendelong, 2020-03-22:
Any reason https://recom-power.com/pdf/Innoline/R-78-1.0.pdf wouldn't be a suitable (superior) replacement for the MCP1826S-3302E-AB?

Looks like the specs are basically identical, but this is a switcher instead of a linear, so it should be more efficient (less heat dissipation, less power input, etc.

I'm looking at possibly adding some hinges and servos to the ESP32 (or ESP8266, haven't fully decided yet) version to create a programmable entrance reducer and also inputs for a load cell to track hive weight (bee and honey production, time to add supers, etc.).

I'm thinking 12, 6, 3, 2, 1 wide doors, facilitating a wide range of entrance sizes.

Curious for any thoughts you might have on this idea.

owendelong, 2020-03-23:
So looking through things, you don't actually seem to use the screw terminals for anything... Is that correct? Is the 3v,A4 screw terminal intended for connecting a load cell?

It also appears you don't actually use the I2C connections for anything. Are they needed? Am I missing something?

Thanks in advance.

BTW, love the bee logo on the board.

hydronics2, 2020-03-23:
Yes! the screw terminals are for adding sensors too. Nothing identified but that's what they are for.

I agree load cells are super important for looking at hive health and activity. 

That's an interesting idea adding retractable entrance reducers. 

you bet, I like the idea of the drop-in-replacement mini switching power supply. Especially  for say reducing parasitic losses (?).

hydronics2, 2020-03-23:
Also, If you are digging into this... I'm missing the jumpers on this CAD version. Do you follow? They are in the gerber files. They jumper the LED resistors. Just a heads up. If you already ordered the board you can easily short the SIP resistors. [That's what I did on my first version. ](https://github.com/hydronics2/2019-easy-bee-counter/blob/master/PCB_files/Eagle_v0.2/shorted_SIP_resistors.png)

owendelong, 2020-03-24:
I haven’t ordered the board… I’ve just started analyzing it for modification (hinges and servos), so no worries there.

I haven’t even ordered the LED/Sensor units yet. (ordered pretty much everything else).

Still trying to figure out how I can “see” the IR LEDs in an all-Apple household.

I’m guessing I’ll end up with a 49th sensor on a separate AT328P with a cheap OLED display or such to use as a “scanner”.

Owen

hydronics2, 2020-03-24:
ha.. good luck. You can also use an old digital camera to see the IR leds.... if you still have one of those.

owendelong, 2020-03-24:
I’ll find something that works… If I do end up building a device and code that can be used to “scan” the LEDs, do you want me to contribute it to the repo?

hydronics2, 2020-03-25:
Sure, if you find it helpful. Thanks

owendelong, 2020-03-28:
Any chance you can provide me the eagle part for your resistor bypass jumpers?
(Either the name of the eagle library and part name, or, a copy of the part)?

I've been looking at the board and I found a few interesting things... I noticed that the sensors were kind of all over the place in the vertical. That seemed to make some routing a bit more challenging, so I aligned them all to 715 and 879, respectively. I also noticed that while the gate headers are on even 500 mil centers (5435, 5935, 6435, ...), the sensors are to the left and right of the 250 mil mark between the headers. I've re-centered those.

I also noticed a few pieces of left over routing cruft (bits of overlapping pieces of traces). Harmless, but created some odd line shapes when zoomed in close.

I'm working on rerouting the board now, hence it would be nice to get the jumpers you used to be able to place them in this process.

hydronics2, 2020-03-30:
Hey there, sorry for the delay getting back to you. I lost my access to Eagle Pro ( left my previous job) so I actually transitioned to kiCAD and added the jumpers in kiCAD..... soooo I don't have the eagle file for that jumper. I"m going to upload the kiCAD files.

owendelong, 2020-03-30:
OK... I used a pair of vias close together to hopefully do the same job. If I end up redoing it, I'll replace those with an 0204 0Ω resistor, which in hindsight is what I probably should have done in the first place).

owendelong, 2020-03-30:
I keep wanting to switch to kicad, but the learning curve seems steep and I have so many parts I've built in Eagle that it's hard to get over the hump.

hydronics2, 2020-04-01:
it wasn't as bad as I thought it would be. What really forced me to try it out was when I needed to make edits to this large board!... There's an import button in Kicad that easily (and seemingly accurate) method to import eagle files.  The other thing that works well on kicad is the 3d rendering.. Check out these renderings.. https://github.com/hydronics2/COVID-19-Airflow-Sensor-AFH55M12

Overall, Eagle has a much smoother interface. There's a lot of choppy zooming in and out in Kicad that is nerve racking...

owendelong, 2020-04-01:
> On Apr 1, 2020, at 11:02 , Thomas <notifications@github.com> wrote:
> 
> 
> it wasn't as bad as I thought it would be. What really forced me to try it out was when I needed to make edits to this large board!... There's an import button in Kicad that easily (and seemingly accurate) method to import eagle files. The other thing that works well on kicad is the 3d rendering.. Check out these renderings.. https://github.com/hydronics2/COVID-19-Airflow-Sensor-AFH55M12 <https://github.com/hydronics2/COVID-19-Airflow-Sensor-AFH55M12>
> 

Hmmm… I’ll have to give it another look… I don’t remember that being there last time I tried it out. Does that import work for library parts or just boards?

I was less worried about importing existing board designs and schematics (I don’t tend to make anything very complex) than I was about getting my library parts over.
> Overall, Eagle has a much smoother interface. There's a lot of choppy zooming in and out in Kicad that is nerve racking…
> 
The good news is that’s probably a combination of CPU power and optimization and among the easier things to fix (vs. fundamental flaws like exorbitant commercial
licensing fees, restrictive licensing, inability to move pin names in library parts, etc.).

Thanks,

Owen

owendelong, 2020-05-05:
So I've had to kind of back-burner this project for a bit. Still making some progress, but it's slow.

Still working in Eagle for now, but I agree those are some nice 3d renderings you pointed out.

One thing that drew me back into Eagle was the Fusion 360 integration (it's now a single license for the previous price of Eagle).

Any chance you could send me the QRE1113 library you used? I don't have it.

owendelong, 2020-05-05:
FYI, In the version I'm modifying, I've rewired the SIP resistors for the sense lines to make the board layout a little easier. (put them in order on the SIP pack to match the sensor placement).

hydronics2, 2020-05-06:
ok, I did a library export using ULP and put the libraries folder in with
the eagle files:
https://github.com/hydronics2/2019-easy-bee-counter/tree/master/PCB_files/Eagle_v0.2
FYI, I'm pretty sure I got those reflective sensors from the standard
Sparkfun library.

owendelong, 2020-05-24:
Here's a Fusion 360 animation of what I have in mind for the reducer doors.

It's still minimal and very crude but I think it's enough to give you the general idea.

If you look at my version of the board, you'll a series of holes along the outside edge (edge away from the hive). The hinge mounting posts fit into those holes. The servos will be mounted to a case that will cover the board. Bell cranks on the servos will push/pull connecting rods which will operate the arms on top of the doors. The door motion is shown in the animation.

http://www.delong.com/Bee%20Counter%20V3.mp4

hydronics2, 2020-05-25:
wowa... that's really neat. very elegant design.

le0pa-RD, 2020-12-08:
> I agree load cells are super important for looking at hive health and activity.

Hello there, I actually made a weight, temperature, and humidity sensors system to monitor hives, it uses an esp8266 to send data to a central raspberry pi over wifi.
Looking forward to implementing a bee counting system to complete my project.
My only doubt is that IR sensors could get propolized and stop working after a while, has it happened to anyone?

hydronics2, 2020-12-08:
From my experience they actually don't propolize over the sensors... they
do propolize around the sides of the sensors, around the leads, on any
voids or holes.