hydronics2/2019-easy-bee-counter #8: I have an idea

m3talover, 2020-05-26:
Hi firstly what a smart and awesome idea for to do this project, congratulations. And you must know i have little knowledge of electronics and their components. i use arduino (or something like that) to do very simple project. i have an idea for improve this project. In mycountry we use color marker to find queen in hives. i think if any chance to add RGB sensors on this gates we get chance to count and know queen behaviours. if you can do it, this made give us to very good data to hive healt, swarn knowledge or use something else for use in bee behaviour researchs.

hydronics2, 2020-05-27:
Hello! yes, I'm currently tinkering with color sensors. If the data looks
good and seems interesting I'll post the results.  If we were trying to
watch out for the Queen we would need a color sensor on each gate.... That
would be a lot of color sensors! My initial interest in color sensors was
to look at pollen color. A lot of bee researchers are interested in pollen.

m3talover, 2020-05-28:
It's interesting to look pollen collor, really good idea too. I'm waiting for your updates.