hydronics2/2019-easy-bee-counter #10: Translate into micropython

jhn4, 2020-06-11:
Hello, i tried to translate your code into micropython, i'm not a vey experienced coder, I dont get it to work really. I'm using the Fipy micro controller from pycom, i don't know if you are familiar with it?
This is the code that i almost copied from your arduino code, it's the same expect how things are written in micropython.
The issue i got is that i don't get anything from the sensors, they're always 0. That makes me think that the issue is in the serial connection.

```
from machine import Pin
import utime
# How many shift register chips are daisy-chained.
NUMBER_OF_SHIFT_CHIPS = 3

# Width of data (how many ext lines)
DATA_WIDTH = NUMBER_OF_SHIFT_CHIPS * 8

# Width of pulse to trigger the shift register to read and latch.
PULSE_WIDTH_USEC = 5

# Optional delay between shift register reads.
POLL_DELAY_MSEC = 1

pinValues = 0
oldPinValues = 0

def read_shift_regs():
    bitVal = 0
    bytesVal = 0
    ploadPin.value(0)
    utime.sleep_us(PULSE_WIDTH_USEC)
    ploadPin.value(1)

    for i in range(DATA_WIDTH):

        bitVal = dataPin.value()
        # Set the corresponding bit in bytesVal.
        print(i,bitVal,bytesVal)
        bytesVal |= (bitVal << ((DATA_WIDTH-1) - i))
        clockPin.value(1)
        utime.sleep_us(PULSE_WIDTH_USEC)
        clockPin.value(0)
    return bytesVal

def display_pin_values():
    print("Pin states: ")
    for i in range(DATA_WIDTH):
        if(pinValues >> i & 1):
            print(" Pin-" ,i ,": HIGH")
        else:
            print(" Pin-" ,i ,": LOW")
    print("\r\n")

#Set up
ploadPin = Pin('P2', mode=Pin.OUT) #Blue
clockPin = Pin('P14', mode=Pin.OUT) #Purple
dataPin = Pin('P4', mode=Pin.IN) #Yellow

clockPin.value(0)
ploadPin.value(1)
pinValues = read_shift_regs()
display_pin_values()
oldPinValues = pinValues
while True:
    pinValues = read_shift_regs()
    if(pinValues != oldPinValues):
        print("Pin value change detected*\r\n")
        print(pinValues)
        display_pin_values()
        oldPinValues = pinValues
    utime.sleep_ms(POLL_DELAY_MSEC)

```

hydronics2, 2020-06-11:
did you start with the example code... test_shift_registers?
https://github.com/hydronics2/2019-easy-bee-counter/blob/master/arduino/test_shift_registers/test_shift_registers.ino

yeah... it looks like Adafruit wrote a little library for the shift out
register 74hc595... It might be helpful to start from this code... buy yeah
writing your own library as a beginner programmer can be a bit daunting!
Maybe post to the arduino forum and see if anyone wants to give you some
pointers? There are lots of helpful people there...

Try posting your question here:
https://forums.adafruit.com/viewforum.php?f=60
If you copy me on your post/answer I'll try to help out too.

hydronics2, 2020-06-11:
here's a link to the 74hc595 library (shift-out) shift register..
https://github.com/adafruit/Adafruit_CircuitPython_74HC595
We need a library for the 74hc165 (shift-in) shift register

jhn4, 2020-06-11:
My code accualy works! The issue was the pload pin was constant 3.3V, so when i switched pins it worked fine!

hydronics2, 2020-07-20:
That's great! Do you have a link to your code?

jhn4, 2020-07-21:
Here is the code, you can post it on your github! I'm not so familiar with Git and you also have the other code so it's better for other people to have all code in the same place!

This code runs on a pycom device, but i think the code will work on any micropython device since the imports are not pycom specific.

from machine import Pin
import utime
# How many shift register chips are daisy-chained.
NUMBER_OF_SHIFT_CHIPS = 3

# Width of data (how many ext lines)
DATA_WIDTH = NUMBER_OF_SHIFT_CHIPS * 8

# Width of pulse to trigger the shift register to read and latch.
PULSE_WIDTH_USEC = 5

# Optional delay between shift register reads.
POLL_DELAY_MSEC = 1

pinValues = 0
oldPinValues = 0

def display_pin_values():
    print("Pin states: ")
    for i in range(24):
        if(pinValues >> i & 1):
            print(" Pin-" , i , ": HIGH")
        else:
            print(" Pin-" , i , ": LOW")
    print("\r\n")

def read_shift_regs():
    bitVal = 0
    bytesVal = 0
    ploadPin.value(0)
    utime.sleep_us(PULSE_WIDTH_USEC)
    ploadPin.value(1)

    for i in range(DATA_WIDTH):

        bitVal = dataPin.value()
        # Set the corresponding bit in bytesVal.
        bytesVal |= (bitVal << ((DATA_WIDTH-1) - i))
        clockPin.value(1)
        utime.sleep_us(PULSE_WIDTH_USEC)
        clockPin.value(0)
    return bytesVal

#Set up
ploadPin = Pin('P23', mode=Pin.OUT) #Blue
dataPin = Pin('P22', mode=Pin.IN) #Yellow
clockPin = Pin('P21', mode=Pin.OUT) #PARPEL

clockPin.value(0)
ploadPin.value(1)

clockPin.value(0)
ploadPin.value(1)

pinValues = read_shift_regs()
display_pin_values()
oldPinValues = pinValues

while True:
    pinValues = read_shift_regs()
    utime.sleep_ms(POLL_DELAY_MSEC)
    pinValues = read_shift_regs()
    utime.sleep_ms(POLL_DELAY_MSEC)
    if(pinValues != oldPinValues):
        pinValues = read_shift_regs()
        if(pinValues != oldPinValues):
            print("Pin value change detected*\r\n")
            display_pin_values()
            oldPinValues = pinValues
        utime.sleep_ms(POLL_DELAY_MSEC)

// Johan

hydronics2, 2020-07-21:
awesome.. updated the code snipit

hydronics2, 2020-07-21:
https://github.com/hydronics2/2019-easy-bee-counter/blob/master/arduino/python.py

closed 2020-07-21 by hydronics2