hydronics2/2019-easy-bee-counter #11: Question

andrijevicnebojsa, 2020-06-12:
Dear Mr. Hudson,
Thank you for this very helpfull project, i ordered everything for both of your bee counters. I have finished this one first and i need to ask you few simple questions. Thank you in advance for answers.
I use adafruit ESP32 and i had to change serial begin to 9600
I. Can i attach sim modul on this esp32 and send data on cloud couse i dont have wireless conection near. 
II. I am getting this results, is this ok or am i missing something

[image]

[image]
[image]
[image]

hydronics2, 2020-06-13:
that looks really good soldering. You can short these pins to get better 'throw' for your IR LEDs.
[image]

hydronics2, 2020-06-13:
As far as the data... the 12 OT 435 means the out sensor at gate 12 was triggered for 435ms. The 12 IT 510 means the IN sensor at gate 12 was triggered for 510 ms. Based on these times and the closness of the triggers the bee counter counted 1 bee coming IN.  later down the serial data shows T 3 1 which means total OUT bees is 3 and total IN bees is 1. (started at the 'out' sensor and ended at the 'in' sensor means the bee was moving IN.... count 1! [image]

andrijevicnebojsa, 2020-06-14:
First I will try sensors in the "field" for a few days without shorting pins, to see results. I will mount it on hive. I will see results and i will give you feedback. But i dont have wifi in the field so for data collecting i must use sim module, but i dont have experiance with connecting sim modul on adafruit.
I am using code that you provide for testing.

hydronics2, 2020-06-14:
how far away is your network?  I've only done a little work with sim
loggers. They are pretty easy, you can just send packets (bee totals) to
the sim logger over serial.

This is a very simple board that logs data locally to an SD card:
https://www.sparkfun.com/products/13712

andrijevicnebojsa, 2020-06-15:
It is very far so i wil try to figure out to connect it with sim module. I have allready made arduino weather station too, and will collect data near hive (humidity (in, out), temp (in, out), UV, wind, rain. I will assemble all this within a few days just to finish code ( i dont have much experiance in arduino codeing), and i will be free to get back to you with some more questions :).

closed 2020-07-20 by hydronics2

andrijevicnebojsa, 2020-07-26:
Hi, sorry for late reply, this is an update. I made some improvements on hive and a central data collection station. I have made two of your projects, one I have already shown you but the other is not working. On first I used adafruit esp32 (I only had one), for the second I had adafruit feather M0 LoRa and adafruit esp8266. I was not able to set any of them to work and I did everything according to your instructions. For LoRa, I chose adafruit feather M0 on arduino ide but it did not want to load the code, and on esp8266 I soldered the shifts register but did not want to accept the code only half of the photoresistor worked. Can you help me what i am doing wrong.
[image]
[image]
[image]
[image]
[image]

andrijevicnebojsa, 2020-07-26:
Can you provide me the code for adafruit LoRa and ESP8266. 
Thank you in advance. This is how hive is looking, i will send you pictures after it is all conected, but just to know it took me a while to finish this up. I am exited and very nervous to get start measuring
[image]

hydronics2, 2020-08-01:
hey there. Sorry for the delay. Did you get the V0 board? or the V1 board?

For the V0 board... you can make it work with the LoRa... if you swap out the shift register chips. Check this out and see if it makes sense. https://github.com/hydronics2/2019-easy-bee-counter/tree/master/instructions/archive

andrijevicnebojsa, 2020-08-02:
I have V1 board.

andrijevicnebojsa, 2020-08-20:
Hi Tomas, i find a solution, i bought another adafruit esp32 feather :).
But now i have one problem when i test gates with paper stick, it shows me  in and out even if i only put papir in. Most of time it detects it like out??
[image]
Did i do something wrong?
Thank you for answer.

andrijevicnebojsa, 2020-08-20:
I also have a problem with Honey bee counter II, I can't make it work. If you allow I will ask a question here.
I can't detect the inputs and outputs with the code you set or I'm not doing something right
[image]
[image]
[image]
Do i have to soldier photoresistor B on gate1 with V2?

andrijevicnebojsa, 2020-08-20:
Do i have to soldier photoresistor B on gate1 with V2?
I have two types of gates but for now none is working, can we make it work

hydronics2, 2020-08-24:
ok... for the EasyBeeCounter... that looks really good. If the shift register code is working.. as you show.. then all the hardware is good.  Looking at the data again... I think it looks good. The information shown is for debugging... so even though it shows information about both gates.. it is not actually counting both gates... 
[image]
Here is what the data is showing...
Gate 6:
Bee is present on the IN sensor for 405ms
Bee is present on the OUT sensor for 420ms
Because the bee was at the IN sensor first and then the OUT sensor... the bee is recorded as leaving the gate. OUT +1
Gate 11:
Bee is present on the OUT sensor for 240ms
Bee is present on the IN sensor for 240ms
Because the bee is present on the OUT sensor first and then the IN sensor... the bee is recorded as moving in through the gate. IN +1
Gate 14:
Bee is present on the OUT sensor for 375ms.
Bee is present on the IN sensor for 390ms.
Because the bee is present on the OUT sensor first and then the IN sensor... the bee is recorded as moving in through the gate. IN +1

Total count for that period is... T,1,2
Bees OUT: 1
Bees IN: 2

hydronics2, 2020-08-24:
I'm sorry I insinuated the data was incorrect at first... now it looks correct.  There are also some other tests... like the time between tripping the in and out sensors has to be reasonable... < 200ms
Also the speed of the bee moving through the gates has to be reasonable... ~<600ms
(I just saw an error in the code where it looks like I use a threshold of 600ms on going IN and 650ms on going OUT.)

andrijevicnebojsa, 2020-08-25:
Do you have maybe solution for Honey bee counter II?
Do you have code for counting?