hydronics2/2019-easy-bee-counter #12: Build Question - 9 pin 22R SIP resistor

kyemacdonald, 2020-07-20:
Hi Thomas,

I am building my first one of these and I couldn't obtain the 10pin 22R SIP resistors locally so ended up with the 9 pin options you have in your alternate parts list.  

Before I solder, could you confirm which pin hole remains unconnected?

Thanks

hydronics2, 2020-07-20:
sure, you need to have the common pin connected... The common pin are
marked with a square on the PCB and also indicated on the SIP with writing
per the cut sheet.

[image: image.png]

hydronics2, 2020-07-20:
[image]

kyemacdonald, 2020-07-21:
Thanks Thomas,

I assumed that was the case but wanted to confirm before I soldered everything up.  It has taken that long for the parts to arrive I didn't want to wreck it!

hydronics2, 2020-07-21:
no sweat. good luck!

closed 2020-07-21 by hydronics2