hydronics2/2019-easy-bee-counter #13: 2019-easy-bee-counter modify with SD and SIM800 bug

rodece, 2020-10-13:
Hi,

very good job!!!!

I am a italian beekeeper technician, and also an electronic one….,  I collaborate with an association of beekeepers, I have realized your beautiful project to monitor the depopulation of hives due to the use of insecticidal products on crops.
As per your project, I used the Feather M0 with SD card as there is no wifi signal for data transfer in the stations in my areas.
I modified the sketch with the data saving function on SD and it works fine, now I wish I could use the data transfer on the thingspeak site using the SIM800L card (I did the tests with Arduino and a DTH22 sensor and it works fine).
I connected the SIM800L on the Rx Tx port of the Feather by activating the serial1,
I then modified the sketch, created by you, to use the SIM800 and when compiling it does not report any errors, the AT commands do not give any errors but the bee count does not work it seems that it is blocked by the sim800 protocol. Commenting on lines 706 to 775 the counter works but clearly not the SIM800L.
Some idea ??
Thank you

[bee_counting_0907_davide_ok_rtcM0_sim800_ver5.zip](https://github.com/hydronics2/2019-easy-bee-counter/files/5370193/bee_counting_0907_davide_ok_rtcM0_sim800_ver5.zip)

hydronics2, 2020-10-13:
hmmmm... do you think it is related to writing to Serial1? try writing to Serial1 in setup and see if that works. Serial1 pins are free and should work.

As an aside, did the SD card work? The MISO, SCK lines are tied up writing to the shift registers. Apparently the shift registers I choose (the ones everyone uses) are not true SPI devices and don't work as expected when you pull the chip select HIGH... So you can't share the SPI with another device (like the SD card)... but maybe it worked for you? [Here's a little more information about this issue... ](https://electronics.stackexchange.com/questions/121249/how-to-put-a-74hc165-on-an-spi-bus)

I would initialize the SIM800L module in setup if possible... Make sure you're not blocking the code and the delay statements are not 'hard delays', that is, write to the module at intervals in line 625, make sense?

`if (millis() -  previousMills > intervallo ) {
Serial1.println("AT");
  delay(1000);

  Serial1.println("AT+CPIN?");
  delay(1000);
....................................
`

rodece, 2020-10-21:
Hi

,
I managed to get it to work partially (but for now it can also be fine ...).
I fixed the sketch by creating different voids, one for the serial, one for the SD and one for the SIM800, then inserting them on the loop from line 568 with common execution time control, currently 2 minutes.
Now it does not count for only 45 seconds during the writing to SD and the subsequent activation of the SIM800 to download data on the site.
More than good !!!!
They advised me to create an array for the section on sending AT commands to the SIM800  ( 705 - 778 )in order to eliminate a good part of the delays.
I'm working on it, even if I have some difficulties ... ...
[bee_counting_1410_ok_SD_sim800_ver5.zip](https://github.com/hydronics2/2019-easy-bee-counter/files/5418470/bee_counting_1410_ok_SD_sim800_ver5.zip)