hydronics2/2019-easy-bee-counter #14: Share HSPI with lora RFM95 module

Christophe45, 2021-01-29:
First many for your project!! it's working very well, I like it. 
Since 1 week I face to an issue, I want to had your bee counter to my system. 
I developed a PCB with an ESP32 wroom 32 with weight sensor, temperature, GPS, an accelerometer and everything uses the LORA system. My Lora module is an RFM95 which uses SPI (VSPI). I tried several things in order to share the VSPI bus as a slave but it doesn't work. I tried using the HSPI bus for your count. Only everything works well but when I put the Lora system nothing works ... Do you have an idea to make everything work? I can share everything with you :)
https://github.com/Christophe45/ESP32-Lorawan
Many thanks in advance for your help :)

hydronics2, 2021-01-29:
Did you see this?
https://github.com/hydronics2/2019-easy-bee-counter/tree/master/instructions/archive

" I made a few mistakes on the first version (v0). The 74hc165 is not a
true SPI device. It doesn't have a tri-state buffer on the MISO...."

Christophe45, 2021-01-29:
Many many thanks Thomas for your answer, I will try this :).

have a nice week-end 

christophe

https://github.com/hydronics2/2019-easy-bee-counter/tree/master/instructions/archive

Christophe45, 2021-01-30:
Hi Thomas,

Now I undersand more, but if I use the 74HC589 I need the re-make a new PCB as the Pins are differents with the 74HC165... :(

If I have to re-make a PCB, I will may be try to use register that using I2C :)

A a nice day

christophe

Le vendredi 29 janvier 2021 à 19:31:31 UTC+1, Blain Christophe <blain.christophe@yahoo.fr> a écrit : 

Many many thanks Thomas for your answer, I will try this :).

have a nice week-end 

christophe

https://github.com/hydronics2/2019-easy-bee-counter/tree/master/instructions/archive

hydronics2, 2021-01-30:
If you are using the PCB v1 then I already made the changes and you can just swap out the shift registers and it will work.
[image]

Christophe45, 2021-02-01:
I have the same PVB V1 my question was link to the register 74HC589 that you  discribe here:

 In order to work with the LORa module you need to swap out the 74hc165 shift registers with true SPI based shift registers... like the 74HC589 referenced in the above stack exchange answer.

The pinout of the 74HC165 is diffent of the 74HC589 so you can't install at the same place?

Many thanks in adanve for your answer :)

Christophe

hydronics2, 2021-02-01:
damn... that right.... the HC595 has a different pinout!... so that's not an easy solution... that's too bad. 

I should have probably designed a tri-state buffer into the design like Paul S. suggests here: 
[image]

https://www.pjrc.com/better-spi-bus-design-in-3-steps/

tri-state buffer: [74AHC1G125](https://www.digikey.com/en/products/detail/diodes-incorporated/74AHC1G125SE-7/2639244)

I've used one of these in a circuit once. Unfortunately there's no through-hole version. You could try to patch one of these in with some small wires and hot glue. It's totally doable but a bit tricky to solder! You'd basically cut the trace on the MISO line and add jumper wires between the pins... and add an enable jumper wire to the OE pin.  

[image]

Per the data sheet, pulling the OE pin high would free up the MISO line for other uses. What do you think?

Christophe45, 2021-02-02:
Many thanks Thomas for your answer :), I will look that ;). But before I will try Bit-banged SPI : https://www.gammon.com.au/forum/?id=11979&fbclid=IwAR0evnHg7Bk4rl7QrX_K8_500TE5-9ZDSsS4beKH7vyye8QyhOgowQofdF8

In order to try with others Pin the SPI :).

I will let's you know :)

christophe 

https://www.pjrc.com/better-spi-bus-design-in-3-steps/

Christophe45, 2021-02-04:
I succeeded with the  Bit-Banged SPI with your sytem and using normal SPI for my LORA RFM95 module :)

Now I will try to make and other PCV with my system ( Weight, temp, GSP, accelerometer and light sensor) :). If you want I can share you my system a this end :)

Best regard,

christophe

Le mardi 2 février 2021 à 08:32:36 UTC+1, Blain Christophe <blain.christophe@yahoo.fr> a écrit : 

Many thanks Thomas for your answer :), I will look that ;). But before I will try Bit-banged SPI : https://www.gammon.com.au/forum/?id=11979&fbclid=IwAR0evnHg7Bk4rl7QrX_K8_500TE5-9ZDSsS4beKH7vyye8QyhOgowQofdF8

In order to try with others Pin the SPI :).

I will let's you know :)

christophe 

https://www.pjrc.com/better-spi-bus-design-in-3-steps/

hydronics2, 2021-02-04:
Great, bit-banging was a good idea.

chinswain, 2024-06-01:
Hi @Christophe45 , do you have a build log of your system? It sounds very interesting - what are you using the accelerometer for?