hydronics2/2019-easy-bee-counter #16: all ir are blinking except gate8

padrian2s, 2021-03-03:
I love this project and I'm trying for the second try to make it works. In the blink test all ir are blinking except gate 8.
On the shift registers test only gate 3 is working and is triggering all the gates together.
Any sugestions on how to check what's going on in the code or on the board?

thanks!

hydronics2, 2021-03-04:
Ok, sounds like you are making good progress. I would check that you have solder on all the pins.  Each gate's LEDs are in series.... if you miss a solder joint, it will show up as the entire gate (both LEDs) not getting power.

hmmm as far as the shift registers... seems like something is definitely throwing everything off. I feel I've ran into this issue before but I cant recall.

padrian2s, 2021-03-04:
Hi Thomas,

Thank you for quick response. I will re-check the soldering part on that gate. I'm using "Adafruit Itsy Bitsy 32u4 - 5V 16MHz"
but also I ordered the Feather ESP32 so I'm trying to get the identical and reduce the root cause of problems.
I'm very curious since I checked the entire board with multimeter:

- input voltage on every pin, leds and shift registers.... all good.
- I also ring some of the circuit paths

My initial tought was that since I'm using different microcontroller maybe shifts registers are not compatibile with SPI but how and why only gate 3 is detect motion and only gate 8 is not working in the IR test.

My PCB was printed in December 2020 at jlcpcb so the missmatch in design is highly improbable. I looked over the code but I can't find anything. 
If you have paid support I will gladly accept this to help me out.

Again, this project is really cool and I appreciate your work and future ideas

padrian2s, 2021-03-05:
Hi Thomas,

Is working! I replace the itsybity with Feather ESP32 and is working just fine!! I can't belive it, by first opensource hardware project is working.

Thank you!!!

closed 2021-03-05 by padrian2s