hydronics2/2019-easy-bee-counter #19: False triggers because of sunlight

kim-ado, 2021-08-11:
I would like to first of all thank you for making this project, it is a brilliant idea!

I recently built this project, and works like a charm inside where there is no sunlight. However, when outside the sensors are all over the place with false triggers constantly. How to mitigate this issue?

hydronics2, 2021-08-11:
I think I've seen that issue when the sun is in low angle positions. So say early morning and the sun is shining into the gates such as,, early morning hive is facing East or late afternoon when hive is facing West. Does that seem to fit your situation? Maybe putting a small planter ~1ft in front of the hive?

kim-ado, 2021-08-11:
Hi and thanks for quick the answer!

The sensors seems to be constantly triggered in the middle of the day (recently tested at 14-16 pm) here in Trondheim Norway. Putting the counter in the shadows seems to mitigate this issue, but it will still be unstable. I've thought about 3D printing a cover of sort with extended roof and/or removing the soldering from the jumpers to decrease throw range of the IR sensors. Any other tips or tricks are greatly appreciated!

hydronics2, 2021-08-13:
I take it the bottom PCB is black, right?... Maybe the sheen on the PCB is reflecting the IR. Maybe try putting some black tape in the gates to better absorb reflecting IR?

kim-ado, 2021-08-24:
It seemed like the landing plate for the bees was reflecting sunlight, putting some black tape on that proved to work! Thanks for the suggestion.

closed 2021-08-24 by kim-ado