hydronics2/2019-easy-bee-counter #20: Suggestion: Make a shorter version with 12 gates instead of 24

oskarscholander, 2022-01-27:
I love this project!

I would suggest adding a short version that supports 12 gates instead of your original 24. Some type of hives here in Sweden doesn't have as wide opening as Langsroth has.

chinswain, 2022-03-20:
+1 for the UK also, 90% of beehives here are National with small entrances (5").

chinswain, 2022-04-16:
Hi all, my PCBs arrived today:

[image]

I'm trying to work out a way to mount it to my hive as it's a little long. Would anyone be able to shorten the PCB design to 34cm or less?

[image]
[image]

oskarscholander, 2022-04-19:
This would be a great improvement to this project. In your case you are quite easily able to cut away some of the box material, but the most common hive entries in Sweden are

oskarscholander, 2022-04-19:
This would be a great improvement to this project. In your case you are quite easily able to cut away some of the box material, but the most common hive entries in Sweden are only 12cm long.

I would love to have the knowledge on how to change the schematic properly but I don't know how :D

chinswain, 2022-04-19:
Unfortunately it is a shared hive, I've been disallowed from cutting it. I'm currently learning Kicad, I think we can modify the design fairly easily.

oskarscholander, 2022-04-19:
Aaah I see.

Haha, I downloaded Kicad right after I wrote the last comment to you! 😂 Let's try to do this!

oskarscholander, 2022-06-07:
@chinswain I wont be able to create a shorter version.

Do u have a spare PCB to sell to a fellow beekeeper in Sweden?

chinswain, 2022-06-08:
I've cut mine and starting building them so no spare.

oskarscholander, 2022-09-21:
@chinswain hope your hives are healthy! How is the bee counter project going?

selmanonal, 2023-02-16:
Is anyone who made shorder version of the pcb?

chinswain, 2024-04-27:
I've started looking at this again.

I cut the PCB as pictured (discarding bottom half) and soldered the components, LED test works:
[image]
[image]

But unfortunately the shift registers do not. I'm not sure why as there are no missing connections that I can see for the 3rd (and now last) 74HC165 caused by the cutting.

Here's the modified shift register test code, I've removed the 4th, 5th and 6th:

```#include <SPI.h>

const int LATCH = A5; 

const byte powerGates1 = 15;

void setup ()
{
  
  SPI.begin();
  SPI.beginTransaction(SPISettings(3000000, MSBFIRST, SPI_MODE2)); //3Mhz
  
  Serial.begin (115200);
  Serial.println ("Begin switch test.");
  pinMode (LATCH, OUTPUT);
  digitalWrite (LATCH, HIGH);

  pinMode (powerGates1, OUTPUT);
  digitalWrite(powerGates1, LOW);
  
}  // end of setup

byte switchBank1;
byte oldSwitchBank1; 
byte switchBank2;
byte oldSwitchBank2; 
byte switchBank3;
byte oldSwitchBank3; 

void loop ()
{
  digitalWrite(powerGates1, HIGH);
  delayMicroseconds(15); //first 24 gates only need 15us while gates closer to the end need ~40us-75us
  
  
  digitalWrite (LATCH, LOW);    // pulse the parallel load latch
  delayMicroseconds(3);
  digitalWrite (LATCH, HIGH);

  delayMicroseconds(3);
  
  digitalWrite(powerGates1, LOW);

  switchBank1 = SPI.transfer (0); //8
  switchBank2 = SPI.transfer (0); //16
  switchBank3 = SPI.transfer (0); //24
  
  byte mask = 1;
  int sensor = 1;

  for (int i = 1; i <= 8; i++)
  {
    if ((switchBank1 & mask) != (oldSwitchBank1 & mask))
      {
      Serial.print ("Switch ");
      Serial.print (sensor);
      Serial.print (" ");
      Serial.println ((switchBank1 & mask) ? "ir triggered" : "none");
      }  // end of bit has changed
    mask <<= 1; 
    sensor++; 
  }  // end of for each bit

  mask = 1;
  sensor = 9;
  for (int i = 1; i <= 8; i++)
  {
    if ((switchBank2 & mask) != (oldSwitchBank2 & mask))
      {
      Serial.print ("Switch ");
      Serial.print (sensor);
      Serial.print (" ");
      Serial.println ((switchBank2 & mask) ? "ir triggered" : "none");
      
      }  // end of bit has changed
    mask <<= 1; 
    sensor++; 
  }  // end of for each bit

  mask = 1;
  sensor = 17;
  for (int i = 1; i <= 8; i++)
  {
    if ((switchBank3 & mask) != (oldSwitchBank3 & mask))
    {
      Serial.print ("Switch ");
      Serial.print (sensor);
      Serial.print (" ");
      Serial.println ((switchBank3 & mask) ? "ir triggered" : "none");
    }  // end of bit has changed
    mask <<= 1;
    sensor++;  
  }  // end of for each bit

  oldSwitchBank1 = switchBank1;
  oldSwitchBank2 = switchBank2;
  oldSwitchBank3 = switchBank3;

  delay (20);   // must be greater than 7.5ms (see notes at top of code)
} ```

chinswain, 2024-04-27:
I just tested using a TV infra red remote control which triggered all of the sensors using the above script so I'm not sure what the issue is, must be the LEDs?

chinswain, 2024-05-01:
The new NDP6020P arrived today, same issue. Does anyone have any ideas?

Bypassing the MOSFET:

[image]

With NDP6020P

[image]

hydronics2, 2024-05-02:
Hey, I'm just trying to catch up... What is the problem? The IR LEDs are not triggering a 3.3V sensor output? Is that correct?

I see you tested the shift registers by shorting 3.3v to the shift register input. That was smart to test. There is also sufficient IR in any semi-direct sun light that will also trigger the sensors and perhaps your IR remote control (I never tried that but cool idea). I like to use a small piece of white paper at a short distance away from the sensor (about 1/4" to 3/8").

Be careful with those LEDs! The LEDs work OK with the 22ohm resistors installed. Good enough to test the functionality of the shift registers. Once the jumpers are soldered, the LEDs should only be allowed to stay ON for 100ms.... otherwise they will burn themselves out.

[image]

chinswain, 2024-05-02:
Hi Thomas,

Thanks for replying and thanks for making the PCB available to the public.

The problem is the LED's are barely lighting up, I cannot work out why. Bypassing the MOSFET pads 2 & 3 proves everything is soldered correctly but using the MOSFET the LED's barely light up at all. From crude testing I  assumed there was an issue with the NDP6020P from china, so ordered a few from Amazon and eBay (It's no longer available from reputable stores).

After many failed attempts at modifying in KiKad we've cut your PCB in half, in theory it should work looking at the schematics. We cut it exactly here:

[image]

Everything works fine, gates respond to an IR remote and IR LED's on cameras trigger them. Any ideas?

hydronics2, 2024-05-02:
Ok, so when you bypass the mosfet (shorting across the mosfet) everything works?  LEDs trigger the sensors?

[image]

FYI, you don't want to bypass the mosfet if you have soldered these jumper pads.... Does that make sense? I'm not sure the documentation emphasized this enough. The LEDs can get damaged if the forward voltage is at 3.3V for any length of time over 100us.
[image]

chinswain, 2024-05-02:
Exactly, via a webcam with the IR filter removed they are very bright, pads are open and when I short the MOSFET pins it's for less than a second.

hydronics2, 2024-05-02:
Ok, if those solder pads are open then you can short the mosfet all day and there is no problem.  

The other mosfet you mentioned, the [NDP6020](https://www.onsemi.com/pdf/datasheet/ndp6020p-d.pdf), is a P-channel mosfet. The one we need is an N-channel mosfet. The generic one I listed the [FQP30N06](https://www.mouser.com/ProductDetail/512-FQP30N06L) is super generic. It should work fine but you can substitute another one as long as it is type N-channel and the threshold voltage is around 2.5V, and the pinout is the same (the ordering of the pins is G, D, S as opposed to D, G, S or something else).  The threshold voltage is the voltage needed to turn it ON.

Try jumping 3.3V to the Gate 'G' pin. This should mimic the way the micro-controller turns the mosfet ON. If that works then you may just have a damaged trace from the micro-controller to the 'G' pin. This is common when you solder/resolder. Its easy to fix by adding a little jumper wire between the micro-controller pin and the 'G' pin. 

[image]

chinswain, 2024-05-02:
jumping 3.3V to the gate produces the same result (very dim LED's). I've checked the traces with a multimeter and can't find any breaks, the 3.3V signal is making it to the gate pin of the MOSFET.

I noticed the drain is only 2v when under load, without the LED's it's 3.3v on a breadboard, if I add a standard LED to the drain and 3V source it drops from 3.3V to 2.6V.

Everything points towards the MOSFET I think? Bizarre as I've tried three some different sources...

ChatGPT suggests [IRF3708PBF ](https://www.mouser.co.uk/ProductDetail/Infineon-Technologies/IRF3708PBF?qs=2r01AXMCG3P2FLBF0Ge6xA%3D%3D)as an alternative, Vgs is 2.8V.

hydronics2, 2024-05-03:
I think both mosfets should work... but sticking with the FQP30N06L for now... 

Can you check the voltage at the Drain pin when the gate is triggered ON. Looking at the FQP30N06L datasheet there should be ~40mV resistance between the drain and the source. If we do our LED math and evaluate the current across 2 LEDs, with forward voltage of 1.25V each, supply voltage of 3.3V, and a 22ohm resistor, our current is about 36ma.

So the voltage drop across the mosfet from drain to source is 36ma x 40mV => 0.001V... so pretty much nothing. Sorry to be rambling on here... I'm just thinking out loud because its been awhile sense I looked at this and to be honest I haven't considered for some time the resistance of an N-channel mosfet. I knew it was small but that is really small. So we should be getting good current passing through this mosfet with little resistance... 

All this is to ask, can you verify again what voltage you are reading at the drain pin when the gate is triggered ON? Could it be you have a short somewhere on your gate pin? Does the gate pin show a resistance of 10k to ground?

[image]

chinswain, 2024-05-04:
Hi Thomas,

It looks like it was the MOSFET, I replaced it with this and it's working perfectly:

IRLB8721PbF
N channel power MOSFET
TO-220 Package
Vds = 30V max
Max current = 62A
Vgs = 2.35V max
Rds_on = as low as 8.7 milliohm - depending on Vgs

https://cdn.shopify.com/s/files/1/0176/3274/files/irlb8721pbf.pdf?v=1646930126

Thanks for replying!

So that confirms it - if people want a shorter version, just cut exactly as shown.

hydronics2, 2024-05-05:
good to hear! I made a note on the readme page

closed 2024-05-05 by hydronics2