hydronics2/2019-easy-bee-counter #25: Bug report: Connection missing for Itsy Bitsy on printed bord

JanolovV, 2024-03-03:
Hi,
I'm working on an Easy Bee Counter and got into trouble in the testing phase, no data from the shift registers. After a long and troublesome (for an amateur like me with limited equipment) I finally found out what is wrong. There is no connection on the printed board between the SCK on my Itsy Bitsy controller and the CLK on the shift registers. So it can't work with any Itsy Bitsy with this design. The clock connection does not go to any other pin on the Itsy Bitsy either.
I checked the corresponding clock connection for the ESP32 controller and there the connection is ok so it's possible to solve the problem with a very short patch.

I'm glad I could figure this out before the bee season starts.

Regards
Jan-Olov

hydronics2, 2024-03-07:
wow that is strange and really unfortunate.  Good job finding it! I tested the board but must have a different gerber file posted. I don't have any PCB tools on my computer right now but I used some online tools....

It looks like the Gerber files do not include the SCK connection for the itsybitsy footprint as you found. You are correct, you can jumper from the ESP32 footprint.

The Eagle schematic shows the SCK pins connected on both footprints on the schematic and board files.
[image]

[image]

It looks like it is missing on the kiCad Schematic and board files. See below missing trace.
[image]

Thanks for finding that. I am going to delete the gerber files and make a note on the kiCad schematic.

JanolovV, 2024-03-08:
It was a bit tricky to troubleshoot. First I suspected the shift registers and desoldered them to test them properly. That was a bit tricky and it would have been a good idea to have them in sockets instead. Now all the shift registers I desoldered was in fact ok but one of my spares was not. So I would strongly recommend using sockets for both the shift registers and for the processor board. It makes troubleshooting so much easier.

I also tested the "photocells" while the shift registers where of the board. A simple program that "blinked" the photocells 3 s on 3 s off which gave me time to check with a multimeter that the signal came through to the right contact on the shift register socket that I welded on and I could use my finger to trigger the detection. I think this might also a better way to test the "photocells" than trying to see the IR light which none of my phones or cameras could.

Kind regards
Jan-Olov

hydronics2, 2024-03-08:
Nice job. Well hopefully now that you have done the work, there'll be less
to troubleshoot for the next person!