hydronics2/2019-easy-bee-counter #26: A comment on the final assembly

JanolovV, 2024-03-15:
After reading earlier discussion about making the passage for the bees narrower I bought lower female headers (Mouser 30483508) about 2 mm lower. After cutting the middle four connectors with a cable cutter I also grinded down the edges with a Dremel. That saved me another mm. I also needed to cut away a small part of the header on the opposite side of the first pull down SIP resistors (and under the screw connectors). One of the resistor legs comes through right where the header is suppose to sit.

Kind regards
Jan-Olov

chinswain, 2024-06-07:
What size is the final gate? Did you reduce the width?
I found some 4.3mm height female headers which would reduce the vertical gate size to 7mm. I think maybe a 3D printed insert to reduce the with to 7mm also would help.

JanolovV, 2024-06-07:
My final gate height is 9 mm and I have not reduced the width. Of course 7 mm is better than 9 but we still have the width 10 mm if you dont do anything there to. 

Having read the problem others have had to reduce errors in the interpretaion of the signals I doubt it is worth the effort. You are likely to still have some errors whatever you do since the design does allow the bees to act randomly. If you had some kind of device that made it impossible for the bees to go back once they entered the gate you would be fine. But I think it's better to accept the design as is and if possible just reducing the gate size a bit to avoid serveral bees at the same time in the gate. If we accept that there will always be errors and that the calculation of bees in and out never will add up correctly. We still get a lot of information about what is going on in our hives. We can see when activity is high and combine it with other measurements like ambient temperature and hive weight to learn more about the little creatures.

Making measurements allways tell you something but not allways what you expect. I measured temperature and humidity in three hives during a winter. I expected it to vary with ambient teperature. They had almost constant temperatures of 10, 15 and 20°C respectively through the whole winter. (I don't remember the humidity but that was also pretty much constant.) The temperature was an effect of how close the sensor was to the globe of bees. So there is very little point in measuring temperature in the hive. The bee thermostat regulates it. If temperature should drop during winter the bees are dead and there is nothing you can do until spring anyway.

Jan-Olov

JanolovV, 2024-06-07:
Thank you for your question!
I commented on Github

Regards
Jan-Olov

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>